shumatech/BOSSA #13: bossac is crashing when uploading or downloading small binaries

elFallino, 2015-06-21:
Using bossac 1.5-arduino on Windows 8.1 64Bit.

When up- or downloading binaries that are to small in size (less than 2305Bytes) bossac is crashing.

Working example:
--------------------------------------------------------------------------
bossac.exe -r0x901 test.bin
Device found on COM4
Atmel SMART device 0x28900960 found

Read 2305 bytes from flash
[==============================] 100% (10/10 pages)
done in 0.284 seconds
--------------------------------------------------------------------------

Crashing example:
--------------------------------------------------------------------------
bossac.exe -r0x900 test.bin
Device found on COM4
Atmel SMART device 0x28900960 found

Read 2304 bytes from flash
[at this point Windows is telling me that bossac is not working correctly anymore]
--------------------------------------------------------------------------

zrecommerce, 2015-06-29:
Can you try using the debug flag? (-d, --debug)

elFallino, 2015-06-29:
Sure.
```
bossac.exe -d -r0x900 test.bin
Trying to connect on COM4
Set binary mode
readWord(addr=0)=0x20000800
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0x28900960
version()=v1.4 Jul 21 2009 11:30:50
chipId=0x28900960
Connected at 921600 baud
Device found on COM4
readWord(addr=0)=0x20000800
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0x28900960
Atmel SMART device 0x28900960 found
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x40)
writeWord(addr=0x20001020,value=0x2000c000)
writeWord(addr=0x400e0a00,value=0x600)

Read 2304 bytes from flash
[at this point Windows is telling me that bossac is not working correctly anymore]
```

nmaron, 2016-06-20:
Same issue on windows 10. I upgraded to arduino ide 1.6.9 because 1.6.8 wouldn't load blink on a DUE board. Now I'm getting the same error as reported above.

cmaglie, 2016-08-30:
Fixed by #20

closed 2016-08-30 by cmaglie