shumatech/BOSSA #15: Bossa 1.2.1 64-bit "Not Responding" on Windows 7

paulrpotts, 2015-09-08:
I have a fresh install of Bossa 1.2.1 on 64-bit Windows 7 (Windows 7 Enterprise, Service Pack 1, Core i5, 12 GB RAM). With no SAM device attached, switching to a serial port listed in the popup, or choosing "Auto-Scan," puts the program immediately into "(Not Responding)" state with the cursor spinning. The program never recovers from this state.

zrecore, 2016-09-19:
@paulrpotts ...is this still an issue with the latest BOSSA release?

paulrpotts, 2016-09-20:
I will give it a shot... I have not thought about this in some time as I'm using Atmel's SAM-BA 2.16 successfully now.

paulrpotts, 2016-09-20:
The latest installer I see on the SourceForge page is still bossa64-1.2.1.msi and yes, I have confirmed that it goes into the "Not Responding" state in exactly the same way.

zrecore, 2016-09-20:
@paulrpotts The most recent BOSSA release is 1.6.1, available at https://github.com/shumatech/BOSSA/releases

paulrpotts, 2016-09-20:
Hmmm... I am not sure what to do with that. I don't see a Windows executable, or installer. It looks like there is a Makefile for use with Mingw but given the dozen different IDEs and compilers I already have installed, I'm not sure I want to try to start figuring out what I need to build this... especially given the way that compilers all like to screw with the common $PATH...

adamfilip, 2016-10-18:
@zrecore where can i download a Windows 64 binary to run?

shumatech, 2017-01-25:
I will be doing a full release of all BOSSA programs including a Windows installer for the 1.8 release.  I plan to incorporate a signed installer to bypass unsigned issues on Windows 7 and later.

rvmicide, 2017-02-23:
Have you updated bin of BOSSA under windows or better can you add section "how to compile for windows" ?
In all case thank's for your work

shumatech, 2017-05-22:
BOSSA 1.8 is released including Windows binaries

closed 2017-05-22 by shumatech