shumatech/BOSSA #18: Arduino due can't exit from bootloader mode often

sined23, 2016-04-03:
Hello! PC is connected to Arduino Due by Native USB. Often after sketch uploading through Native USB or when I plug usb cable or after OS restarted (power restart), Arduino Due still keep in Bootloader mode and usb communication with PC doesn't work and sketch doesn't work.
After hardware reset of Arduino Due it become normal.

lsusb | grep boot
Bus 001 Device 020: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader

I run manually bossac's command to reset Arduino Due but it doesn't help:

~/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino$ ./bossac -p ttyACM0 -R

lsusb | grep boot
Bus 001 Device 020: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader

Why can't Bossac reset CPU? Is this bug or need send some additional command to CPU to reset?

BTW: Arduino Due can recieve new sketches but can't be reseted by Bossac.

zrecore, 2016-09-09:
@sined23 ...do you have verbose debugging enabled in the Arduino preferences? If this is still happening, please attach the output here (or at least, reply with it using code markdown)

shumatech, 2017-01-25:
BOSSA is a general purpose utility for Atmel devices so Arduino-specific issues belong on the Arduino forums.

closed 2017-01-25 by shumatech