shumatech/BOSSA #19: Wish: SAML21 Xplained Pro

Mecrisp, 2016-04-05:
A wish: May I use BOSSA for Atmel SAM L21 Xplained Pro in future ?

zrecore, 2016-08-11:
Conceivably, this should be possible. I'd be happy to do it, but I don't currently have any SAM L21 chips (or SAM L21 Xplained Pro boards). It's a matter of adding the correct constants to Devices.h, updating FlashFactory.cpp,  and making sure we have the correct page size, page count, etc (Nvm config).

shumatech, 2017-01-25:
Due to existing devices getting broken by adding newer device support (looking at you SAMD), I will not be able to add new device families to BOSSA unless I have a board to test it on for future releases.  If anyone wants to donate a board with a device family that I do not have, then please let me know.

closed 2017-01-25 by shumatech