shumatech/BOSSA #27: Fast-forward master branch into arduino branch

cmaglie, 2016-08-30:
Hi @shumatech 

I'm wondering if it still makes sense to have two different branches, one from the original bossac (`master` branch) and the other the arduino version (`arduino` branch)?

From the git log I see that the `master` branch` could be fast-forwarded to the current `arduino` branch, would you mind if I do that, and let everyone continue to work on the master branch?

projectgus, 2016-08-30:
(Following on from #2). Wow, blast from the past! If there is a current version then I have ATSAM4L support here, from back in 2013: https://github.com/shumatech/BOSSA/compare/master...projectgus:atsam4l

... that I'd be happy to send a PR for.

shumatech, 2016-09-01:
Hi @cmaglie 

I'm fine with merging the arduino branch into master so long as there are no Arduino-specific functionality as a default.  All board-specific functions should be enabled by command line switches.  I haven't used any Atmel (or is it Microchip now?) processors in designs for several years now so I'm not in a good position to test any BOSSA changes.  If someone is willing to take this up and upgrade the GUI for the latest changes then that would be great.

cmaglie, 2016-09-09:
Wonderful, thanks @shumatech !

commit 134544d8 (2016-09-13): Fixed readme and started 1.7.0. Added CHANGELOG.

Fix #27
closed 2016-09-09 by cmaglie