shumatech/BOSSA #28: Add support for SAMD11 chips

todbot, 2016-10-13:
Looks like mattairtech already has done most of the work:
https://github.com/mattairtech/BOSSA/commit/fcf8a701af69d2821c84c025c72b36ecb6d0efd2
and it would be really awesome if this was in the official version.

(and thanks for BOSSA, it's awesome)

shumatech, 2017-01-25:
You're welcome.  Due to instances of existing devices getting broken when new devices are added, I will be unable to add support for new devices unless I have a board with that family to do regression testing on prior to each release.  If anyone wants to donate a board to the project, then please let me know.

closed 2017-01-25 by shumatech

todbot, 2017-01-25:
Hi @shumatech, 
Email me with a shipping address (tod at todbot dot com) and I'll get a few boards out to you.
Cheers!