shumatech/BOSSA #33: XXXX_BOOTLOADER_SIZE as a parameter

AloyseTech, 2016-12-02:
Hi,
Working on the SAMD/R chip series, it is possible to make bootloaders bigger/smaller than 8kB. Would it be possible to pass the bootloader size or application start address as a parameter?
Something like this for example : 
bossac -e0x8000 -w0x8000 -v0x8000 -b image.bin
or
bossac --startaddr=0x8000 -e -w -v -b image.bin

shumatech, 2017-01-25:
In BOSSA 1.8, I added an offset feature where you can specify an offset to start a read, verify, or write operation.  So to skip the first 16KB, you would execute:

bossac -o 0x4000 -w -v -b image.bin

Note that the SAMD does not currently support auto-erase while programming so the above will not work for a SAMD.  I can add auto-erase support to the SAMD but unfortunately, I do not have a board with a SAMD on it to develop it on.

cw2, 2017-06-20:
> In BOSSA 1.8, I added an offset feature where you can specify an offset to start a read, verify, or write operation. So to skip the first 16KB, you would execute:

Hm, does the offset feature work also in the opposite direction, i.e. when the bootloader is smaller than 8 KB (4 KB)? It seems the bootloader size is hardcoded in [NvmFlash.cpp](https://github.com/shumatech/BOSSA/blob/master/src/NvmFlash.cpp#L101) and so is the flash start address in [Device.cpp](https://github.com/shumatech/BOSSA/blob/master/src/Device.cpp#L325).

How about completely abandoning hardcoded assumptions about the bootloader, use the whole flash starting at 0x0000 and leave to the user to provide correct offset where necessary? He/she can lock the bootloader region to prevent accidental rewrite, if needed... (I can imagine BOSSA to actually read the lock bits).

vshymanskyy, 2018-02-09:
Hi guys. I'd like to bring this issue back to the table. Are there any updates?

shumatech, 2018-03-14:
I removed all knowledge of the bootloader size from bossa.  It is not really bossa's problem how the user partitions their flash.  To support bootloaders, I added two features.  The erase command -e now respects the offset -o so you can specify the bootloader size with that.  Also, I added support for auto-erase to the SAMD21 and SAMD51 so you don't need to explicitly erase.  For example, to erase, write, and verify a device with a 8KB bootloader, run:

bossac -p /dev/ttyACM0 -e -w -v -o0x2000 flash.bin

Note this will erase to the end of flash, to auto-erase only the pages written, omit the -e.

closed 2018-03-14 by shumatech