shumatech/BOSSA #36: Pages to erase

dipasqua, 2017-01-23:
Hi,
it could be very useful to have an additional parameter for the erase command: the total pages to erase. Indeed, it can be used to keep data stored in NVM with the EEPROM emulation driver.

shumatech, 2017-01-25:
The EFC and EEFC flash controllers on all non-SAMD Atmel devices only supports a full chip erase or an auto-erase before writing so this cannot be done.  It might be possible on a SAMD though I don't know much about them.

closed 2017-01-25 by shumatech

resdevok, 2017-01-27:
Hi shumatech,
thanks so much for your reply.
Yes on a SAMD is possible: in fact we modified ATSAMD21G18A_FLASH_PAGES from 4096 to 4032 to keep data stored at the and of the flash and it properly works.

SietseAchterop, 2020-09-23:
The nRF52640 can erase on a page base, and it would indeed nice to be able to do partial erase!