shumatech/BOSSA #40: Unable to flash SAMD21 via UART

samyk, 2017-03-04:
Hello!

I'm attempting to flash firmware (compiled via platformio) onto an ATSAMD21G18 via UART on Raspberry Pi 3. The Pi is connected to the following ATSAM's pins:
- PA15 (for SAM-BA bootloader monitor mode) [rpi pin 11]
- PA22 (UART TXD for SAM-BA UART programming) [rpi pin 10]
- PA23 (UART RXD for SAM-BA UART programming) [rpi pin 8]
- RESET pin [rpi pin 7]
- SWDIO (for writing bootloader/debugging via openocd) [rpi pin 18]
- SWCLK (for writing bootloader/debugging via openocd) [rpi pin 22]
- GND
- 3V3 for power (Pi powering ATSAM)

I tested and was able to program the ATSAM over USB using bossac, but am not able to do so over UART. UART is enabled on the Pi3 as I have /boot/config.txt contain enable_uart=1 and core_freq=250.

I can use openocd to successfully write the bootloader, and I've tried using three different SAM-BA bootloaders (first two from the ZIP file https://github.com/femtoduino/femto-usb/blob/master/Atmel_files/SAM-BA_Bootloader_and_Programmer-revision_A-Updated_10-2014/Atmel-42366-SAM-BA-Bootloader-for-SAM-D21_ApplicationNote_AT07175.zip): samd21g18a/samd21_sam_ba_uart.hex, samd21g18a/samd21_sam_ba_both_interfaces.hex and https://github.com/arduino/ArduinoCore-samd/blob/master/bootloaders/zero/samd21_sam_ba.bin.

Here's an example (output changes over time):
```
# currently using samd21_sam_ba_uart.hex
pi@rpidev:~/bootloader $ alias gp='gpio write'
pi@rpidev:~/bootloader $ alias slp='sleep 0.3'
pi@rpidev:~/bootloader $ # on pi, reset atsam (pull RESET pin low)
pi@rpidev:~/bootloader $ alias atreset='gp 7 0 ; slp ; gp 7 1'
pi@rpidev:~/bootloader $ atreset
pi@rpidev:~/bootloader $ sudo bossac --info --debug --port ttyS0 --erase --write --verify --reset -U false blinkfirmware.bin
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000280
readWord(addr=0x400e0740)=0x170000
chipId=0x170000
Unsupported processor
No device found on ttyS0
```

Any thoughts or suggestions?

PS, I'm happy to create a board for the chip I'm using and contribute it for the project! Thanks!

samyk, 2017-03-04:
This is interesting. I mentioned I was able to use bossac over USB -- that was from another machine (macOS) but onto the same board (ATSAM). When I attempt to use bossac on the Pi, I get the following:
```
pi@rpidev:~/bootloader $ ls /dev/ttyA*
/dev/ttyAMA0
pi@rpidev:~/bootloader $ atreset
pi@rpidev:~/bootloader $ ls /dev/ttyA*
/dev/ttyACM0  /dev/ttyAMA0
pi@rpidev:~/bootloader $ sudo stty -F /dev/ttyACM0 1200
pi@rpidev:~/bootloader $ sudo bossac --info --debug --port ttyACM0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0x400e0740)=0x60000
chipId=0x060000
Unsupported processor
Send auto-baud
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0
chipId=00000000
Unsupported processor
No device found on ttyACM0
```

samyk, 2017-03-04:
Okay, I've resolved the issue regarding flashing via USB on the Pi and have that working now, however I am still not able to flash via UART (I was using the Pi's version of bossac which is outdated [1.3a], now upgraded to 1.7.0)

Any suggestions? Thanks!

```
pi@rpidev:~/bootloader $ sudo stty -F /dev/ttyS0 1200
pi@rpidev:~/bootloader $ sudo ~/bossac --info --debug --port ttyS0 --erase --write --verify --reset -U false blinkfirmware.bin
Send auto-baud
Set binary mode
No device found on ttyS0
pi@rpidev:~/bootloader $ sudo stty -F /dev/ttyS0 115200
pi@rpidev:~/bootloader $ sudo ~/bossac --info --debug --port ttyS0 --erase --write --verify --reset -U false blinkfirmware.bin
Send auto-baud
Set binary mode
No device found on ttyS0
pi@rpidev:~/bootloader $ atreset
pi@rpidev:~/bootloader $ sudo ~/bossac --info --debug --port ttyS0 --erase --write --verify --reset -U false blinkfirmware.bin
Send auto-baud
Set binary mode
No device found on ttyS0
```

I can get into SAM-BA monitor mode

samyk, 2017-03-04:
Some more progress, now actually able to communicate with the chip. I was able to do this by changing the bootloader to `samd21g18a/samd21_sam_ba_both_interfaces.hex`

```
root@rpidev:/home/pi/bootloader# alias gp='gpio write'
root@rpidev:/home/pi/bootloader# alias slp='sleep 0.3'
root@rpidev:/home/pi/bootloader# # on pi, reset atsam (pull RESET pin low)
root@rpidev:/home/pi/bootloader# alias atreset='gp 7 0 ; slp ; gp 7 1'
root@rpidev:/home/pi/bootloader# # on pi, go into atsam bootloader (pull PA15 low during reset)
root@rpidev:/home/pi/bootloader# alias atbootloader='gp 0 0 ; slp ; atreset ; slp ; gp 0 1'
root@rpidev:/home/pi/bootloader#
root@rpidev:/home/pi/bootloader# atreset
root@rpidev:/home/pi/bootloader# ../bossac --info --debug --port ttyS0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000470
readWord(addr=0xe000ed00)=0x410cc601
No device found on ttyS0
root@rpidev:/home/pi/bootloader# ../bossac --info --debug --port ttyS0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000470
No device found on ttyS0
root@rpidev:/home/pi/bootloader# ../bossac --info --debug --port ttyS0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000470
No device found on ttyS0
root@rpidev:/home/pi/bootloader# atbootloader
root@rpidev:/home/pi/bootloader# ../bossac --info --debug --port ttyS0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000470
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305

SAM-BA operation failed
root@rpidev:/home/pi/bootloader# ../bossac --info --debug --port ttyS0 --erase --write --verify --reset -U true blinkfirmware.bin
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x20000470
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v1.1 Aug 11 2014 20:42:19
chipId=0x10010005
Connected at 115200 baud
readWord(addr=0)=0x20000470
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)

SAM-BA operation failed
```

shumatech, 2017-05-21:
SAMD21 is not officially supported since I do not have a dev board to test it with.  If anyone wants to donate a SAMD21 board to the BOSSA project, I'll officially support it and add it to my regression tests done prior to release.

samyk, 2017-05-22:
I'd be happy to donate a board to support the project. Let me know where you'd like it sent. You can reach me at git@samy.pl.

joaofl, 2017-05-22:
I would appreciate the support for the SAMD21 as well. 

I've been using this very accessible dev board:
https://www.aliexpress.com/item/Crowduino-M0-SD-Board-for-Arduino-UNO-Micro-Controller-Board-DIY-Kit-Free-Shipping/32712839065.html?spm=2114.13010608.0.0.ViAqDC

closed 2018-08-25 by shumatech

samyk, 2019-04-19:
Some notes I left myself in case it helps others (doing this from a Raspberry Pi):

* You must first disable BLE on Pi (dtoverlay=pi3-disable-bt enable_uart=1).
* I've also had to run this twice in some cases.
* You may need to adjust `serial0` to `serial1` or greater if there are multiple serial devices
* You may need to adjust the `serial` in `serial0` to `ttyAMA`
* You may need to remove `console=` section and set `core_freq=250` in /boot/config.txt, then run
  * `sudo systemctl stop serial-getty@ttyS0.service`
  * `sudo systemctl disable serial-getty@ttyS0.service`

First you must get the chip in bootloader mode by holding PA15 low and pulling the reset low.
*"The bootloader will check PA15 on reset to determine if the bootloader monitor shall start. This feature allows the end user to reprogram the device even if the application is corrupted or una- ble to start the SAM-BA monitor"*

```sh
alias gp='gpio write'
gpio mode 1 out
gpio mode 7 out
gp 1 1
gp 7 1
alias slp='sleep 0.3'

# on pi, reset atsam (pull RESET pin low)
alias atreset='gp 1 0 ; slp ; gp 1 1'

# on pi, go into atsam bootloader (pull PA15 low during reset)
alias atbootloader='gp 7 0 ; slp ; atreset ; slp ; gp 7 1'

# go into sam-ba monitor mode when using arduino sam-ba bootloader
atreset ; atreset

# go into bootloader mode when using femto (real) sam-ba bootloader
atbootloader # doesn't always work?

sleep 1

sudo bossac --info --debug --port serial0 --erase --write --verify --reset -U false blinkfirmware.bin
```