shumatech/BOSSA #43: compilation error: 'unique_ptr' in namespace 'std'

rabryan, 2017-03-17:
With gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4), I'm getting the compiler error below.  Fix is to add '-std=c++11' to COMMON_CXFLAGS in Makefile. I'll submit a pull request shortly.

CPP APPLET src/WordCopyArm.cpp
In file included from src/Samba.h:38:0,
                 from src/Applet.h:34,
                 from src/WordCopyApplet.h:32,
                 from src/WordCopyArm.cpp:3:
src/SerialPort.h:74:18: error: ‘unique_ptr’ in namespace ‘std’ does not name a template type
     typedef std::unique_ptr<SerialPort> Ptr;
                  ^
In file included from src/Applet.h:34:0,
                 from src/WordCopyApplet.h:32,
                 from src/WordCopyArm.cpp:3:
src/Samba.h:54:30: error: ‘SerialPort::Ptr’ has not been declared
     bool connect(SerialPort::Ptr port, int bps = 115200);
                              ^
src/Samba.h:97:17: error: ‘Ptr’ in ‘class SerialPort’ does not name a type
     SerialPort::Ptr _port;
                 ^
src/Samba.h: In member function ‘const SerialPort& Samba::getSerialPort()’:
src/Samba.h:74:49: error: ‘_port’ was not declared in this scope
     const SerialPort& getSerialPort() { return *_port; }
                                                 ^

joaofl, 2017-03-28:
Getting the same... Did yo manage to solve it?

rabryan, 2017-03-29:
Just add '-std=c++11' to COMMON_CXFLAGS in Makefile.  Take a look at PR #44

joaofl, 2017-04-04:
@rabryan 
Thanks. I added that flag, and It actually starts to compile now, but now I'm getting a second error:

```
src/BossaForm.h:11:24: fatal error: wx/artprov.h: No such file or directory
 #include <wx/artprov.h>
```

However, I have already installed libwxgtk2.8-dev and libwxgtk3.0-dev, and other packages more. But the error persists... Did you have this problem?

Did you have this issue? Remember how you solved it?

Thanks

rabryan, 2017-04-05:
I just modified line 7 in the Makefile to specify WXVERSION=2.8.  Not a clean, commitable solution obviously

closed 2017-05-21 by shumatech

guysoft, 2023-12-03:
Solution also works for wx version 3.2 on ubuntu 23.04