shumatech/BOSSA #50: Forcing Bootloader mode on SAMD21 to avoid having to press the reset button

deladriere, 2017-06-13:
I am using BOSSA 1.8 on Mac
It works greats but it needs the board to be placed in bootloader mode by pressing the reset button twice. My board is placed in an enclosure with no direct access to the reset button.
Could you add a line like
` stty -f /dev/cu.usbmodemfa121 1200;stty stop /dev/cu.usbmodemfa121`
just before the bossac command. 
This would force the board to enter into bootloader mode. I have tried the command from the terminal and it allows BOSSA to detect the board.
This is also the trick (opening closing the port at 1200 bps)used by the Arduino IDE to enter into bootloader mode (I found this trick here: [https://arduino.stackexchange.com/questions/1160/why-does-command-line-bossa-not-recognize-arduino-due]

casundra, 2017-06-13:
I would love this too!

shumatech, 2018-03-18:
This should be in the current master using the -a switch.  I haven't tested it since I do not have a SAMD Arduino.

closed 2018-03-18 by shumatech