shumatech/BOSSA #53: BOSSA not working in MacOS Sierra

WZ9V, 2017-07-06:
MacOS Sierra 10.12.5, BOSSA GUI looks like it starts but then just exits without doing anything.  this is with the supplied version 1.8 DMG from the releases.

Hardware is Early 2015 MacBook 12 using external hub to Duet Wifi

Not sure what is needed to recompile on MacOS so have not tried that option.

odboy, 2018-01-15:
Hi WZ9Z, 
Do you already solve this problem?
It's same that after I mistakenly selected Bluetooth USB port instead J-Link USB port。

shumatech, 2018-02-28:
I'll be doing the 1.9 release on High Sierra so hopefully that will fix things.

shumatech, 2018-03-18:
I'm running 10.13.3 HS with no issues right now.  I'm using homebrew wxmac 3.0.3.1_1.

rob-miller, 2018-03-20:
Not sure if the OP was trying to connect to a DuetWifi or PanelDUE, the latter being the device this software is specified for upgrading.  For the PanelDUE, the USB port is not enabled by the pre-loaded firmware.  Thus the port will not appear in the drop-down list (and nothing will work) until the 'erase' button has been pressed and the device subsequently reset.  Before I learned this, I selected the bluetooth port while trying things as odboy referenced, and got it (Bossac v1.8) into the loop described where it starts and immediately exits until the relevant files under ~/Library/ are deleted.  I confirm that v1.9 compiles easily with homebrew wxmac, but upon learning about needing to erase I went back to the mac air with Bossac 1.8 on it and was successful with that.

WZ9V, 2018-03-25:
Duet WiFi but I’ve moved on from MacOS and used a Linux box instead.

closed 2018-03-25 by WZ9V

cchaloin, 2018-03-27:
Hi,
when do you plan to release an OSX binary for 1.9 ?
Thank you