shumatech/BOSSA #57: No device found on /dev/ttyACM0 Arduino Due.

girish946, 2017-09-01:
Hello,
I'm having an arduino due, I'm trying to upload zephyr RTOS onto it.
when I do

```bash 
./bossac -i -d -p /dev/ttyACM0 -U true -e -w -v -b  ~/git/zephyr/samples/hello_world/outdir/arduino_due/zephyr.bin -R
```

I'm getting the output as

```
Set binary mode
Send auto-baud
Set binary mode
No device found on /dev/ttyACM0
```
Am I missing something?
can anyone help me with this?

ThomasChr, 2017-09-06:
Check if the device is listed unter 'lsusb'.
If so, try resetting and starting bossac again. You might do this a few times, but finally it works after a few tries.

prasimix, 2017-09-19:
I have the same problem working with Arduino Due. Tried it many times but got the same error. My linux is Ubuntu 16.04LTS, `bossac` version 1.7.0 and `lsusb` returns:

```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 046d:081b Logitech, Inc. Webcam C310
Bus 003 Device 028: ID 2341:003e Arduino SA 
Bus 003 Device 011: ID 13fd:0840 Initio Corporation INIC-1618L SATA
Bus 003 Device 004: ID 054c:0104 Sony Corp. 
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 008: ID 8087:07da Intel Corp. 
Bus 003 Device 007: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Arduino IDE doing some tricks with usb/serial port reset and `bossac` works there, but I need solution for people who are not familiar with Arduino IDE and just want to upload file.

ThomasChr, 2017-09-19:
You need to push ERASE for a few seconds while the Due is running. After that push RESET for a short period, wait a second, and then you should have no problems to use BOSSAC.

Oh, and use the following command line:

bossac -p ttyACM0 --usb-port=true -e -w -v -b ./firmware.bin

Let me know the outcome!

prasimix, 2017-09-19:
I tried `./bossac -p ttyACM0 --usb-port=true -e -w -v -b /tmp/arduino_build_768174/eez_psu_sketch.ino.bin` but it returns:
```
bossac: unrecognized option '--usb-port=true'
Try 'bossac -h' or 'bossac --help' for more information
```
This suggestion with ERASE sounds good, but even if works it's unfortunately not practical in my case because that key is not easily accessible since due is plugged in in but bigger "shield" and placed inside metal enclosure (link on this [picture](http://www.envox.hr/eez/images/arduino_shield_cables2.jpg)).

ThomasChr, 2017-09-19:
Older Versions of Bossac use '-U true' instead of '--usb-port=true', could you try that?
And If we've got that working we can try the auto erase... theres a trick for that.

girish946, 2017-09-19:
For me also `Arduino` shows up in `lsusb` output when it is connected to programming port but not when it is connected to the native port.
I also have write permission on the port ttyUSB0.

kevans91, 2017-09-19:
I have an open PR that would add a flag to help this situation: #45 (Add 1200bps erase flag)

ThomasChr, 2017-09-19:
@kevans91 Yes, thats exactly what I'm talking about. A flag would be a good idea!

prasimix, 2017-09-19:
Wow this one seems to work! I'm still not sure that combination of ERASE and RESET keys is the right one. Since ERASE key is not accessible in my case, a solution that include port reset as kevans91 suggested will be nice.

kevans91, 2017-09-19:
Yeah, the `bossac` flags can vary, what really matters is the 1200bps open/close prior to that to put the Due into the right prerequisite state for BOSSA to work with it.

ThomasChr, 2017-09-19:
You can't talk to the program code of the Due. You need to erase it first. After erase the bootloader starts and this is the one you can talk to.

The ATMEL16U2 Chip on the Due does the communication handling. It has the capability to erase the due and start the Bootloader. It does that when you open it with 1200bps first. That's the trick Arduino uses so that you don't need to Push the Erase-Button.

It should be possible to do that opening with 1200 baud manually and the PR from kevans91 has a view Infos about that.

prasimix, 2017-09-19:
Ok, but I'm using so-called Native USB port and it works quite nice. It's not connected to 16U2, or it is?

shumatech, 2018-03-18:
The Arduino reset is in the latest master.

closed 2018-03-18 by shumatech

dav322, 2023-02-21:
add this line first
stty -F /dev/ttyACM0 1200