shumatech/BOSSA #60: Adding support to ATSAMD21E15A

MartinBloedorn, 2017-11-22:
Hello, 

I'm using an ATSAMD21E15A, with 32k of Flash and 4k RAM - less than the usual 256k/32k combo seen on Arduinos. I've made some slight modifications on the [Arduino Zero bootloader](https://github.com/arduino/ArduinoCore-samd/tree/master/bootloaders/zero) to fit my current setup, and it seems to be working correctly. 

I'm compiling _bossac_ from source successfully, and since there seems to be no out-of-the box support for the particular device I'm using, I've added the proper entry to `Device.cpp`, under the `No CHIPID devices` section:

``` cpp
    case 0x1001020d:
        _family = FAMILY_SAMD21;
        flashPtr = new NvmFlash( _samba, "ATSAMD21x15", 0x2000,  512, 64, 1, 16, 0x20000000, 0x20001000, 0x41004000, true ) ;
        break;
``` 
I can sucessfully connect to the device, fetch its ID, and perform erase operations. However, when I try to write to it, the operation hangs, as shown below:

``` 
$ bossac.exe -i -d -p COM7 -e -w blinky.bin --reset
Set binary mode
version()=v2.0 [Arduino:XYZ] Nov 22 2017 02:32:26
Connected at 921600 baud
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x1001020d
write(addr=0x20000000,size=0x34)
writeWord(addr=0x20000030,value=0x10)
writeWord(addr=0x20000020,value=0x20001000)
version()=v2.0 [Arduino:XYZ] Nov 22 2017 02:32:26
readWord(addr=0x41004018)=0
readWord(addr=0x40000834)=0x7000a
readWord(addr=0x40000834)=0x7000a
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
Device       : ATSAMD21x15
Version      : v2.0 [Arduino:XYZ] Nov 22 2017 02:32:26
Address      : 0x2000
Pages        : 384
Page Size    : 64 bytes
Total Size   : 24KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Chip Erase   : true
Fast Write   : true
Fast Verify  : true
Erase flash
chipErase(addr=0x2000)

Done in 0.084 seconds
Write 6432 bytes to flash (101 pages)
[                              ] 0% (0/101 pages)write(addr=0x20000034,size=0x1000)
``` 
I'm pretty sure that the values for `user` and `stack` (respectively set as `0x20000000` and `0x20001000`) are incorrect, but I'm unsure about what to use. If I set them both to be `0x00002000`, the write operation is completed, but the verification fails, and the device crashes on reboot.

I feel like I'm missing something fairly obvious. Could somebody point me in any direction? 

Thanks!

MartinBloedorn, 2017-11-23:
Adding a bit more information to the mix. I've created a dummy 144-byte test binary, as follows:

``` 
Offset(h) 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

00000000  00 11 22 33 44 55 66 77 88 99 AA BB CC DD EE FF  .."3DUfwˆ™ª»ÌÝîÿ
00000010  00 11 22 33 44 55 66 77 88 99 AA BB CC DD EE FF  .."3DUfwˆ™ª»ÌÝîÿ
00000020  00 11 22 33 44 55 66 77 88 99 AA BB CC DD EE FF  .."3DUfwˆ™ª»ÌÝîÿ
00000030  00 11 22 33 44 55 66 77 88 99 AA BB CC DD EE FF  .."3DUfwˆ™ª»ÌÝîÿ
00000040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000070  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000080  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
``` 

I wrote it to the device using `bossac -i -d -p COM7 -e -w test.bin`. I then read the data back, with `bossac -i -d -p COM7 -r0x100 testread.bin`, which yielded:

``` 
Offset(h) 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

00000000  00 00 02 01 00 40 02 60 08 89 08 80 00 18 00 2A  .....@.`.‰.€...*
00000010  00 11 00 00 00 00 00 51 00 08 80 02 40 45 C0 46  .......Q..€.@EÀF
00000020  00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00  ... ............
00000030  00 00 00 00 00 11 22 33 00 11 22 33 88 99 AA BB  ......"3.."3ˆ™ª»
00000040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000070  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000080  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000A0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000B0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
000000C0  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
000000D0  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
000000E0  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
000000F0  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
``` 
This looks nothing like the supplied binary. The result is the same under Windows and Linux. I still get the feeling that I'm making a crass mistake here. Any tips are welcome.

aksonlyaks, 2017-12-22:
Dear MartinBloedorn,

Are you able to find out the issue? I'm facing the similar issue with SAMD21E16A.

Thanks and Regards
Abhishek

MartinBloedorn, 2017-12-22:
Hi @aksonlyaks, 

Unfortunately I haven't been able to work around the issue yet. I've tried a couple more changes in the `Devices.cpp`,  to no avail. 
In the end, I swapped out my SAMD21E15A for a SAMD21E18A, which is supported out-of-the box by BOSSA. However, I'm still interested in knowing how to properly support the SAMD21E15 (or E16). Let me know if you happen to make any progresses in that direction. 

Regards,

aksonlyaks, 2018-01-05:
Dear Martin,

I'm able to resolve the issue for ATSAMD21E16A.

I have changed the buffer address value to 
#define ATSAMD21E16A_BUFFER_ADDR             (0x20000800ul).    // Earlier the value was 0x20001000ul

I hope this will help you to work with E16A. I guess with E15A we need to change the buffer address value to little lower than default. 

Let me know if it works for you.

Thanks and Regards
Abhishek

shumatech, 2018-03-18:
The latest master should have support for both the E15 and E16 though I don't have any to test with.  I positioned the applet address and stack to be within their SRAM spaces but I don't know if they conflict with the bootloader.  For normal read/erase/verify, the applet is not used but modifying the other flags needs the applet.

closed 2018-03-18 by shumatech

sabas1080, 2018-06-01:
I have the same problem, what tests should I do to help?

wero1414, 2018-06-01:
Here i have a SAMD21E15B and i have the same problem when i try to upload the code with the newer version compiled in my computer

qbolsee, 2022-05-01:
I'm late to the party but I think I have some insight on this. The `write()` hangs only occur when writing large blocks, so one workaround is reducing the block size to e.g. 512 for SAMD21E15 variants. I'm working on an outdated fork but you can have a look at my patch if that's any help:
https://github.com/qbolsee/BOSSA/commit/7750ebcf5d3d058881fb1176201fd72e517cbd53

In my case, I didn't need to modify the buffer addresses to make it work with a SAMD21E15B.