shumatech/BOSSA #64: Bossa 1.8 x86 running on Win XP gives "......local hard drive" message

Riggar, 2018-02-15:
on installation message box shows "Installation directory must be on a local hard drive"

changing drive / names etc always gives the above message

running under cmd prompt admin with/
msiexec \i <path> <with or without \q and any following parameter 

still gives the same message

shumatech, 2018-03-18:
Sorry, but Bossa no longer supports Windows XP.  I only test and target Windows 7 and 10.

closed 2018-03-18 by shumatech

Riggar, 2018-03-18:
> <author@noreply.github.com>
> "......local hard drive" message (#64)

ok thanks

Rigby

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

gvanem, 2023-10-25:
This is still an issue on Win-10:
[image]

OMG!