shumatech/BOSSA #71: Cannot compile with mingw in Windows 10

freemovers, 2018-03-28:
I designed a board with the ATSAME54N20A and I like to use the latest BOSSA 1.9. Unfortunately no binary is created yet, and I am stuck with compiling it with mingw. Do you have any instructions on how to compile this. Currently the following messages appear:

C:\msys32\home\BOSSA>mingw32-make
wx-config Warning: Multiple compiled configurations of wxWidgets have been detected.
Using first detected version by default.

Please use the --wxcfg flag (as in wx-config --wxcfg=gcc_dll\mswud)
or set the environment variable WXCFG (as in WXCFG=gcc_dll\mswud)
to specify which configuration exactly you want to use.
wx-config Warning: Multiple compiled configurations of wxWidgets have been detected.
Using first detected version by default.

Please use the --wxcfg flag (as in wx-config --wxcfg=gcc_dll\mswud)
or set the environment variable WXCFG (as in WXCFG=gcc_dll\mswud)
to specify which configuration exactly you want to use.
CPP APPLET src/WordCopyArm.cpp
In file included from src/Samba.h:38:0,
                 from src/Applet.h:34,
                 from src/WordCopyApplet.h:32,
                 from src/WordCopyArm.cpp:3:
src/SerialPort.h:76:18: error: 'unique_ptr' in namespace 'std' does not name a template type
     typedef std::unique_ptr<SerialPort> Ptr;
                  ^
In file included from src/Applet.h:34:0,
                 from src/WordCopyApplet.h:32,
                 from src/WordCopyArm.cpp:3:
src/Samba.h:55:30: error: 'SerialPort::Ptr' has not been declared
     bool connect(SerialPort::Ptr port, int bps = 115200);
                              ^
src/Samba.h:99:17: error: 'Ptr' in 'class SerialPort' does not name a type
     SerialPort::Ptr _port;
                 ^
src/Samba.h: In member function 'const SerialPort& Samba::getSerialPort()':
src/Samba.h:75:49: error: '_port' was not declared in this scope
     const SerialPort& getSerialPort() { return *_port; }
                                                 ^
Makefile:248: recipe for target 'obj/WordCopyArm.o' failed
mingw32-make: *** [obj/WordCopyArm.o] Error 1

freemovers, 2018-03-28:
I changed to Cygwin, but still have a similar issue. I added COMMON_CXXFLAGS=-std=c++11 in the makefile already per one of the previous posts. Mingw had no issues with the build after that change, but didn't create any binary files. See the new errors below. 

make all 
Building file: ../src/Applet.cpp
Invoking: Cygwin C++ Compiler
g++ -O2 -g -Wall -c -fmessage-length=0 -MMD -MP -MF"src/Applet.d" -MT"src/Applet.o" -o "src/Applet.o" "../src/Applet.cpp"
In file included from ../src/Samba.h:38:0,
                 from ../src/Applet.h:34,
                 from ../src/Applet.cpp:29:
../src/SerialPort.h:76:18: error: 'unique_ptr' in namespace 'std' does not name a template type
     typedef std::unique_ptr<SerialPort> Ptr;
                  ^
In file included from ../src/Applet.h:34:0,
                 from ../src/Applet.cpp:29:
../src/Samba.h:55:30: error: 'SerialPort::Ptr' has not been declared
     bool connect(SerialPort::Ptr port, int bps = 115200);
                              ^
../src/Samba.h:99:17: error: 'Ptr' in 'class SerialPort' does not name a type
     SerialPort::Ptr _port;
                 ^
../src/Samba.h: In member function 'const SerialPort& Samba::getSerialPort()':
../src/Samba.h:75:49: error: '_port' was not declared in this scope
     const SerialPort& getSerialPort() { return *_port; }
                                                 ^
make: *** [src/subdir.mk:121: src/Applet.o] Error 1

13:58:43 Build Finished (took 344ms)

filipkotoucek, 2018-05-02:
Exactly same problem here. I'm using mingw (gcc --version 5.1.0)

freemovers, 2018-05-02:
I ended up just compiling the BOSSAC.EXE from Mattairtech in MSYS with MINGW32, just use the  BOSSA/arduino/make_package.sh. Using BOSSAC in Arduino to upload sketches to a board with SAME54 now.

shumatech, 2018-08-25:
I use msys2 with mingw32 7.3.0 to build the releases.  I configure wxWidgets with:

configure --disable-shared --enable-unicode --with-zlib=builtin --disable-debug