shumatech/BOSSA #76: Any ideas on how to install on High Sierra?

MGonzalez506, 2018-05-29:
Hi, do someone knows how to install it on high Sierra? I have a problem to put bossac and bossach into usr/bin folder because they are just readable.

Thanks! :)

RemcoVerrips, 2018-06-12:
Since MacOS sierra (i thought) u can't use thoose folders anymore because its protected
You can however disable System Integrity Protection

You can disable System Integrity Protection by reboot and press  ```CMR + R``` and start recovery mode, open terminal and type:
> csrutil disable
> reboot

when you restart you will have permission (with root) to change the directory

shumatech, 2018-08-25:
The Mac installer for 1.9.1 release will put them in /usr/local/bin.

sslupsky, 2018-11-07:
I have bossac and bossash installed in /usr/local/bin.  However, I am getting an error when I launch the bossa app:

BOSSA cannot be opened because of a problem.
Check with the developer to make sure BOSSA works with this version of macOS. You may need to reinstall the application. Be sure to install any available updates for the application and macOS.

I am running 10.13.6

Here is the crash report:

`Process:               bossa [69152]
Path:                  /Applications/BOSSA.app/Contents/MacOS/bossa
Identifier:            ShumaTech.BOSSA
Version:               1.0 (1.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           bossa [69152]
User ID:               501

Date/Time:             2018-11-07 16:27:31.928 -0700
OS Version:            Mac OS X 10.13.6 (17G65)
Report Version:        12
Anonymous UUID:        48EEE68D-A8A6-77D4-ED29-7885081922DE

Time Awake Since Boot: 280000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/opt/jpeg/lib/libjpeg.9.dylib
  Referenced from: /Applications/BOSSA.app/Contents/MacOS/bossa
  Reason: Incompatible library version: bossa requires version 13.0.0 or later, but libjpeg.9.dylib provides version 12.0.0

Binary Images:
       0x109f2f000 -        0x10a253ff7 +ShumaTech.BOSSA (1.0 - 1.1) <2250E76C-5EC7-38F8-92B7-593771778389> /Applications/BOSSA.app/Contents/MacOS/bossa
       0x10a368000 -        0x10a38bffb +libpng16.16.dylib (0) <44F2FE8A-980E-335E-A6F9-F6633924FD5E> /usr/local/opt/libpng/lib/libpng16.16.dylib
       0x1170db000 -        0x117125acf  dyld (551.4) <8A72DE9C-A136-3506-AA02-4BA2B82DCAF3> /usr/lib/dyld
    0x7fff47ca2000 -     0x7fff47f30ff7  com.apple.audio.toolbox.AudioToolbox (1.14 - 1.14) <E0B8B5D8-80A0-308B-ABD6-F8612102B5D8> /System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox
    0x7fff48604000 -     0x7fff48604fff  com.apple.Carbon (158 - 158) <F8B370D9-2103-3276-821D-ACC756167F86> /System/Library/Frameworks/Carbon.framework/Versions/A/Carbon
    0x7fff48b16000 -     0x7fff48b16fff  com.apple.Cocoa (6.11 - 22) <78E6C28E-4308-3D10-AD14-0CBCF6789B3F> /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa
    0x7fff4bb2c000 -     0x7fff4bbc7fff  com.apple.framework.IOKit (2.0.2 - 1445.71.1) <2EA4F383-CAA9-3AF0-99C5-90C22ADAA6B6> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff53878000 -     0x7fff53887ff3  com.apple.opengl (16.7.4 - 16.7.4) <9BDE8FF9-5418-3C70-8D1C-09656884CE48> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff6eed2000 -     0x7fff6eed3ffb  libSystem.B.dylib (1252.50.4) <CD555F3B-FDDB-35E5-A2FB-FBBF3D62031A> /usr/lib/libSystem.B.dylib

Model: MacBookPro10,1, BootROM MBP101.00F6.B00, 4 processors, Intel Core i7, 2.7 GHz, 16 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.190.18.1a3)
Bluetooth: Version 6.0.7f10, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en4
Network Service: Wi-Fi, AirPort, en0
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM768E, 751.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5`

Fulg, 2018-12-29:
If you have brew installed you will run into a similar problem as @sslupsky on Mojave (wrong dylib versions).

You can work around it with `brew install wxwidgets`, this will install compatible versions of libpng/libjpeg/libtiff in `/usr/local/opt` and will allow the GUI version of BOSSA to launch correctly on Mojave.

You should probably add your dependent dylibs into the BOSSA application package to solve this issue for good. I think the most popular answer to [this StackOverflow question](https://stackoverflow.com/questions/4677044/how-to-use-dylib-in-mac-os-x-c) will be useful.

Johncoffee2017, 2019-06-25:
@Fulg 
Worked like a charm! Thanks for your suggestion :-)