shumatech/BOSSA #77: Binaries for v1.9

taggie, 2018-06-12:
Would it be possible to provide the (Windows) binaries for version 1.9?
Thanks!

Tomibubi, 2018-06-13:
I would also very appreciate binaries of the last version for Windows.
Thank you.

freemovers, 2018-06-16:
I was never able to compile the BOSSA GUI, but I have a binary of BOSSAC for Arduino, supporting SAMD51 and SAME54. Let me know if that works for you and I will try to get in on Github.

hackwerken, 2018-06-29:
Hi, I was wondering if it is possible to provide binaries for Windows and Mac of the Bossa master (b176eeef918fc810045c832348590595120187b4)?

Thanks in advance!

shumatech, 2018-08-25:
I released binaries for Mac and Windows for 1.9.1.  However, my code signing certificate expired so I cannot sign them.  Windows will most likely show a warning when installing the driver.