shumatech/BOSSA #79: "Removed knowledge of the bootloader size from BOSSA" commit is dangerously compatibility-breaking

aswild, 2018-06-16:
https://github.com/shumatech/BOSSA/commit/dbdd088909c57bdf9aee191720bf6c6159217e22 is great, but also *dangerous* to Arduino SAMx21 users who don't understand its implications.

The first time I ran my "make upload" target, which doesn't specify `--offset` and worked on v1.8, I bricked my chip by erasing the bootloader code.  I have an SWD programmer so it was easy enough to re-flash the bootloader, but others users may not be so fortunate.

There's also a flags incompatibility issue:

Version | --offset=0x0000 (default) | --offset=0x2000
--- | --- | ---
1.8 | Correct, writes to 0x2000 | Incorrect, writes to 0x4000 (but bootloader still works)
1.9 | Dangerous, writes to 0x0000 and kills the bootloader | Correct, writes to 0x2000

I don't have any great ideas for handling the changes to --offset, but I think it could be beneficial to print a warning and require confirmation (or an explicit `--offset=0` argument) before erasing/writing anything at address 0.

deladriere, 2018-06-16:
Thanks ! I know understand what happened to my SAMD21 !

fedorcomander, 2018-07-20:
does bossa always flashes from 0x2000 by default? i am using 4kb bootloader....

aswild, 2018-07-20:
The default flash address (for SAMx21 chips) changed from 0x2000 (8k) to 0x0000 in version 1.9. If you need to flash your application at 0x1000 (4k) then you can do that with `--offset=0x1000` on BOSSA 1.9.

fedorcomander, 2018-07-20:
Took me 2 hours to realise that… ;) Just switched to UF2 bootloader with some changes, works way easier for me ;) No tool required for update ;)

matt448, 2018-08-21:
I just ran into the same thing. Luckily the fuses were set to prevent overwriting the boot loader. The guide I was following didn't specify to use an offset and I now realize that guide was written based on bossa 1.8 which assumes a default of 0x2000. I think @aswild suggestions are good. Maybe just getting rid of a default value requiring an offset would be best.

shumatech, 2018-08-25:
I'm fine with requiring that it be explicitly specified for non-ROM bootloaders like the SAMD's.

tuupola, 2018-08-26:
I think this also [affects SAMx51](https://github.com/adafruit/uf2-samdx1/issues/29).

deladriere, 2018-08-28:
@shumatech how to set the offset in the  bossa GUI ?
It keeps saying  :  "**Error** Flash offset is invalid" whatever I type: 0x2000, 2000 or 8192 ?

deladriere, 2018-11-15:
@shumatech any updates on how to set the offset in the GUI ?

LumitoLuma, 2021-06-22:
Hey, I've made a fork of this repo (https://github.com/LumitoLuma/BOSSA) that aims to solve this problem without loosing the new BOSSA functionality (1.9.1).

I have tested it with my SAMD21G18A and works fine.