shumatech/BOSSA #80: Add support for ATSAMD21E15B

sabas1080, 2018-06-20:
Hi @shumatech  

I am trying to program an ATSAMD21E15B and I get an error, is the chip supported?

How can I help to add support?

Thanks

sabas1080, 2018-06-28:
I get the next error
`Set binary mode
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011427
version()=v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
chipId=0x10010027
Connected at 921600 baud
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011427
Atmel SMART device 0x10010027 found
write(addr=0x20000800,size=0x34)
writeWord(addr=0x20000830,value=0x10)
writeWord(addr=0x20000820,value=0x20001000)
Device       : ATSAMD21E15B
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011427
Chip ID      : 10010027
version()=v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
Version      : v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
Address      : 8192
Pages        : 384
Page Size    : 64 bytes
Total Size   : 24KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.112 seconds

Write 9880 bytes to flash (155 pages)
write(addr=0x20000c00,size=0x1000)
writeBuffer(scr_addr=0x20000c00, dst_addr=0x2000, size=0x1000)`

sabas1080, 2018-06-28:
```./bossac -d --port=cu.usbmodem1421 -e -w -v -i -U -o 0x2000 /var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_48499/Blink.ino.bin -R
Set binary mode
version()=v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
Connected at 921600 baud
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011427
write(addr=0x20000800,size=0x34)
writeWord(addr=0x20000830,value=0x10)
writeWord(addr=0x20000820,value=0x20001000)
version()=v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
readWord(addr=0x41004018)=0
readByte(addr=0x804001)=0xc7
readByte(addr=0x804001)=0xc7
readByte(addr=0x804006)=0xff
readByte(addr=0x804007)=0xff
Device       : ATSAMD21x15
Version      : v2.0 [Arduino:XYZ] Jun 27 2018 12:18:39
Address      : 0x0
Pages        : 512
Page Size    : 64 bytes
Total Size   : 32KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Erase flash
chipErase(addr=0x2000)

Done in 0.112 seconds
Write 9880 bytes to flash (155 pages)
[                              ] 0% (0/155 pages)write(addr=0x20000834,size=0x1000)
writeBuffer(scr_addr=0x20000834, dst_addr=0x2000, size=0x1000)```

sabas1080, 2018-06-29:
I had to comment on the line
https://github.com/shumatech/BOSSA/blob/master/src/Samba.cpp#L123
to work correctly

excsniper, 2018-07-03:
Above is a quick workaround for a bug that I suspect is hanging the flashing process. BOSSA assumes a `writeBufferSize` of 4096B but the chip only has that much ram in total. Relevant lines
https://github.com/shumatech/BOSSA/blob/d480a1ed34f4d08ac4822fe3aa5d1149b53b40c8/src/Samba.h#L85
https://github.com/shumatech/BOSSA/blob/d480a1ed34f4d08ac4822fe3aa5d1149b53b40c8/src/Samba.h#L89
If the sizes are suppose to be the same then may I also recommend refactoring them into a constant.

Gadgeteering, 2019-10-02:
I have had the same is with a SAMR21S16 I reduce the buffer size to 2048 and it worked #110