shumatech/BOSSA #81: Installing 1.8 on Debian?

jhaupt, 2018-07-06:
No configure script in the tarball nor instructions for installing. 

Binaries would be equivalently good :)

aswild, 2018-07-16:
You don't need a configure script, it's just a flat Makefile. Run "make" and then manually install the `bin/bossa*` binaries to `/usr/local/bin` or wherever else you want to install.

thehans, 2018-08-02:
A list of dependencies would be nice and even if the instructions are simply "run make", its better to be explicit than to have everyone guessing what to do.  

I had to install `libwxgtk3.0-dev` for the makefile to run.  Not sure if there were other deps that I just happened to already have installed.

aswild, 2018-08-02:
I think wxgtk and readline are the only dependencies, or at least that's all I have in my [Arch PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=bossa-git)

Nightbr, 2019-02-18:
Hey, I had to install wxgtk and realine like @aswild said:
`sudo apt install libreadline-dev libwxgtk3.0-dev`

It could be great to add this in some install documentation or juste compile some binaries for linux!

Thanks for your work!

twitchyliquid64, 2019-09-12:
I've whipped up this script to automatically build 1.9.1 from source and install it.

https://gist.github.com/twitchyliquid64/72a0ef3d96ad12a61d3f82dfd4367b5e

Should work on all debian-based systems.

paulalting, 2019-10-08:
Yes, I concur, it makes for a better install if build instructions are explicit as @thehans mentions, rather than leaving it up to users to figure it out.

Once I found this post here, I was easily able to get the install correct and things working as they should.