shumatech/BOSSA #91: Build setup on windows10

tecnosel, 2019-02-24:
Hello, 

I need add new deviced Atsamd21j17d in bossac for custom board.
Can someone send me a folder that I can compile in windows for me to modify the device?
I can not compile bossac in windows.

Thanks a lot

beikeland, 2020-05-02:
see https://github.com/shumatech/BOSSA/issues/71#issuecomment-415997591

although I was not successful with current mingw, will update if i find a workaround or how to install the specific version.

marshaul, 2021-03-18:
This took me quite a while to figure out (and I was only able to build the bossac target).

You'll need to install MSYS2 with MINGW32, then from an MINGW32 shell you'll need to install gcc and mingw-w64-i686-wxWidgets. 

I probably forgot something else I did, but if you can run 'make' within mingw32 and tell me what errors you get, I can probably tell you what else you need.

casundra, 2021-10-18:
@marshaul and @shumatech - I'm trying to compile BOSSA to support the SAMD21G17D chip.  It's the latest silicon rev and we couldn't get the A version (thanks chip shortage!) which is already supported by BOSSA.  I've installed MSYS2 with MINGW32, installed gcc, and used pacman to "install" wxWidgets ( pacman -S mingw-w64-i686-wxWidgets ).  I have _not_ compiled my own version of wxWidgets - using MINGW32 and running make is new to me, so I'm a little lost despite having read a ton of forum posts on the topic.  These are the errors I'm getting and I'm wondering if you could help?

make: wx-config: No such file or directory
make: wx-config: No such file or directory
make: git: No such file or directory
CPP BOSSA src/BossaForm.cpp
In file included from src/BossaForm.cpp:8:
src/BossaForm.h:11:10: fatal error: wx/artprov.h: No such file or directory
   11 | #include <wx/artprov.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:260: obj/BossaForm.o] Error 1

lncoll, 2021-11-13:
You need to install wxWidgets, here you have a guide helped me.
https://jrcrulida.blogspot.com/2017/02/compiling-and-installing-wxwidgets-310.html