shumatech/BOSSA #96: Flashing fails with this compiled version while 1.7 is doing fine.

pjalocha, 2019-04-13:


pjalocha, 2019-04-13:
1.7 supplied with Arduino 1.8.9 works just fine, but when I compile and run this version here it fails as below.

The command given is like this:

bossac --port=ttyACM0 -i -U -e -w -R -v <compiled-binary>.bin

and the relevant part of the reply:

Done in 0.740 seconds
Write 49716 bytes to flash (777 pages)
[==============================] 100% (777/777 pages)
Done in 0.255 seconds
Verify 49716 bytes of flash
[==============================] 100% (777/777 pages)
Verify failed
Page errors: 128
Byte errors: 7956

RatGit, 2019-05-01:
I have the same issue I think...

I've compiled versions 1.7.0, 1.8 and 1.9.1 on a Raspberry Pi 3B+ which I use to program a SparkFun SAMD21 Mini Breakout board, (ie. an Arduino Zero clone).

Both 1.8 and 1.9.1 fail in the same way, (see below), whereas 1.7.0 works? One thing I've noticed is that 1.7.0 successfully detects the USB port and connects at the higher speed, (_Connected at 921600 baud_), whereas the later versions don't:
> sudo bossac -i -d --port=ttyACM0 --usb-port=true -e -w -v -R '/home/programmer/Sampler_Controller.bin'
> Send auto-baud
> Set binary mode
> version()=v2.0 [Arduino:XYZ] Dec 20 2016 15:36:39
> Connected at 115200 baud
> readWord(addr=0)=0x20007ffc
> readWord(addr=0xe000ed00)=0x410cc601
> readWord(addr=0x41002018)=0x10010305
> write(addr=0x20004000,size=0x34)
> 
> SAM-BA operation failed

I also find that I have to run the following command beforehand to ensure that bossac will work. (The SAMD21 board also occasionally needs to be unplugged and re-inserted).
```
sudo stty -F /dev/ttyACM0 speed 1200 cs8 -cstopb -parenb
```
If I don't do this I get the following error:
> Set binary mode
> Send auto-baud
> Set binary mode
> No device found on ttyACM0

galak, 2019-05-01:
Did some bisecting and got this commit causing issues:

commit ee5a651aacaf2c3fbcd539789574618a425d1a06 (refs/bisect/bad)
Author: Scott Shumate <scott@esquilo.io>
Date:   Sun May 21 09:58:01 2017 -0500

    Refactored to encapsulate all device-specific knowledge into the Device
    class including flash creation to prevent Samba from being a dumping
    ground for miscellaneous device features.

cmaglie, 2019-07-10:
@RatGit 
the semantic of `-U` parameter has been changed from 1.7.
You may want to try to replace `--usb-port=true` or `-U true` to just `-U`

cmaglie, 2019-07-10:
@pjalocha @RatGit 
try also to add `-o 0x2000` to the command line (I suppose you're running a SAMD21 based board), previously the offset to skip the bootloader size was built-in in bossac, now you must specify it via command line.

nzmichaelh, 2020-06-28:
@RatGit re: the board needing to be unplugged and then plugged back in, I see the same due to a race between the 'stty' call, the board detaching, and Bossa attaching.  If the board hasn't finished restarting then Linux puts the board on ttyACM1 instead of ttyACM0

rzr, 2022-08-22:
I confirm zephyr's version is working fine on arduino_nano_33_ble

I can provide some traces if it helps

```
/usr/local/bin/bossac-1.9.1-14-g3532de8  --version
Basic Open Source SAM-BA Application (BOSSA) Version 1.9.1-14-g3532de8

/usr/local/bin/bossac-1.9.1-17-g89f3556 --version
Basic Open Source SAM-BA Application (BOSSA) Version 1.9.1-17-g89f3556

```