shumatech/BOSSA #104: MacOS - Need to upgrade libtiff

gotfredsen, 2019-08-10:
When I tried to run BOSSA, it just exited without any messages or errors. When I then ran it from the command line, I got this result:
` 
/Applications/BOSSA.app/Contents/MacOS/bossa 
dyld: Library not loaded: /usr/local/opt/libtiff/lib/libtiff.5.dylib
  Referenced from: /Applications/BOSSA.app/Contents/MacOS/bossa
  Reason: Incompatible library version: bossa requires version 9.0.0 or later, but libtiff.5.dylib provides version 8.0.0
Abort trap: 6
`

I then upgraded the libtiff like this:
`brew upgrade libtiff`

And it worked again!

nathan-embedism, 2019-10-02:
This also worked for me - thank you so much!

kr15uk, 2019-12-16:
On clean copy of Catalina I had to brew install libpng libjpeg and libtiff

finson, 2020-03-10:
I'm also working on a relatively clean copy of Catalina 10.15.3 on which I've installed the Zephyr development environment. I got the same sort of Dyld error, only libtiff was not just outdated, it was "image not found" altogether.  I used brew install on all three libs which added libtiff and found libpng and libjpeg already installed.  Bossa now runs.