shumatech/BOSSA #107: Overwrite Bootloader

Gadgeteering, 2019-09-15:
When I try to write a file without setting the offset (in OSX) the bootloader is corrupted and the board does not response any more.

 ### With -o Offset at 0x2000

 bossac -o 0X2000 -i -d --port=cu.usbmodem14201 -U -i -e -w -v  sketch_sep15a.ino.bin -R 
Set binary mode
version()=v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
Connected at 921600 baud
readWord(addr=0)=0x20001ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x1001031e
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x10)
writeWord(addr=0x20001020,value=0x20002000)
version()=v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
Device       : ATSAMR21x16
Version      : v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
Address      : 0x0
Pages        : 1024
Page Size    : 64 bytes
Total Size   : 64KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Erase flash
chipErase(addr=0x2000)

Done in 0.190 seconds
Write 3800 bytes to flash (60 pages)
[                              ] 0% (0/60 pages)write(addr=0x20001034,size=0xf00)
writeBuffer(scr_addr=0x20001034, dst_addr=0x2000, size=0xf00)
[==============================] 100% (60/60 pages)
Done in 0.023 seconds

### Without -o Offset

bossac -i -d --port=cu.usbmodem14201 -U -i -e -w -v sketch_sep15a.ino.bin -R 
Set binary mode
version()=v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
Connected at 921600 baud
readWord(addr=0)=0x20001ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x1001031e
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x10)
writeWord(addr=0x20001020,value=0x20002000)
version()=v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
readWord(addr=0x41004018)=0
readByte(addr=0x804001)=0xc7
readByte(addr=0x804001)=0xc7
readByte(addr=0x804006)=0xff
readByte(addr=0x804007)=0xff
Device       : ATSAMR21x16
Version      : v2.0 [Arduino:XYZ] Sep 15 2019 08:29:47
Address      : 0x0
Pages        : 1024
Page Size    : 64 bytes
Total Size   : 64KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : true
BOR          : true
Erase flash
chipErase(addr=0)

Done in 5.002 seconds
Write 3800 bytes to flash (60 pages)
[                              ] 0% (0/60 pages)write(addr=0x20001034,size=0xf00)

SAM-BA operation failed

As you can see the chipErase addr=0

Has anybody else experienced the same problem

deladriere, 2019-09-16:
If I understand it well you need to use the offset to preserve the bootloader.
BTW I am trying to set the offset with the GUI version on OSX. Are you successful with the GUI too?

Gadgeteering, 2019-09-16:
The issue is not isolated to OSX, I have tried the windows version and it has the same fault. If I set the offset to 0x2000 in windows,OSX and command line, it programs the board correctly. 

It seems that the offset default settings were removed at this commit [https://github.com/shumatech/BOSSA/commit/dbdd088909c57bdf9aee191720bf6c6159217e22#diff-4bc76381991166f0b7d7a4a0a9116fde](url) 
This seems to me as a backward step.

Gadgeteering, 2019-10-02:
Duplicate issue #79