shumatech/BOSSA #114: Are ATSAMD21E17D and ATSAMD21E17B supported?

tinkerBOY-git, 2019-12-17:
I have just tested it with ATSAMD21E17A and it seems to work just fine but i'm wondering if ATSAMD21E17D and ATSAMD21E17B are also supported?

tinkerBOY-git, 2020-01-04:
Bootloader works fine on ATSAMD21E15B but I cannot get BOSSA 1.7.0-mattairtech-3 to upload a binary file. It hangs/stuck on when it's time to "write".

```
D:\BOOTLOADERS\BOSSA 1.7.0-mattairtech-3>bossac -i -d --port=COM27 -U true -i -e -w -v Blink.ino.ATSAMD21E15B.bin -R
Set binary mode
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011527
version()=v2.0 [Arduino:XYZ] Dec 11 2019 22:16:43
chipId=0x10010027
Connected at 921600 baud
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011527
Atmel SMART device 0x10010027 found
write(addr=0x20000800,size=0x34)
writeWord(addr=0x20000830,value=0x10)
writeWord(addr=0x20000820,value=0x20001000)
Device       : ATSAMD21E15B
readWord(addr=0)=0x20000ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10011527
Chip ID      : 10010027
version()=v2.0 [Arduino:XYZ] Dec 11 2019 22:16:43
Version      : v2.0 [Arduino:XYZ] Dec 11 2019 22:16:43
Address      : 8192
Pages        : 384
Page Size    : 64 bytes
Total Size   : 24KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.113 seconds

Write 11040 bytes to flash (173 pages)
write(addr=0x20000c00,size=0x1000)
```

mistoll, 2020-03-12:
SAMD21E17D support was added in https://github.com/shumatech/BOSSA/pull/124