shumatech/BOSSA #119: Bossa 1.9 and 1.8 crash when trying to write

DTS188, 2020-02-10:
This happens on v1.8 and 1.91. As soon as I hit write, Bossa closes or crashes. I need a fix fast.

rabryan, 2020-02-10:
How fast and how much you willing to pay for this fix?

- Richard

DTS188, 2020-02-10:
your kidding

rabryan, 2020-02-10:
No, no I’m not.  Though I’m not the creator of this software, I do know that it is free (as in freedom) software which gives you (for free) the ability to use and fix this software yourself.

I am a professional software engineer with 15 years of experience who is able to fix this bug that, as you indicated, is a big issue for you.  I was simply offering to help you.  And my first step in helping you was to ask a straightforward question to learn more about your issue.

- Richard

DTS188, 2020-02-10:
That wasn't question to learn more about the problem. That was a question that appeared to be a scam for money.
I am not willing to pay anything for this. I have another solution. Send the product back to the manufacturer and have them re-program the device. Something I would rather not do, but that doesn't cost me anything.

rabryan, 2020-02-10:
Ok.  Glad to hear you have a solution then so the issue can be considered resolved.

- Richard

DTS188, 2020-02-10:
No, I have a secondary solution. I would still like to know why this happens.
It's not up to you to decide if this is resolved.

rabryan, 2020-02-10:
True.  But based on the information you have provided no person on this planet can actually know why the crash you are claiming has occurred. 

- Richard

DTS188, 2020-02-10:
Problem solved. Used a different programmer that works.