shumatech/BOSSA #122: Flashing SAME53N20

rbickle, 2020-02-19:
Using BOSSAC to flash SAME53N20 device vis USB and Atmel SAM-BA monitor. The utility communicates with the device, but verify fails. When the flash is read back, it is still erased. I have created files showing the commands issued and the debug responses of each step. I do not know how to interpret the debug data. Thanks.

[e53n20.zip](https://github.com/shumatech/BOSSA/files/4222654/e53n20.zip)

xeno27, 2020-02-20:
maybe same thing as by the SAMC21 (need at least 6ms to erase single page).

you can try my wait ready work around [Pull Request](https://github.com/shumatech/BOSSA/pull/123)

rbickle, 2020-02-20:
Thanks for the info. I’m not sure how to implement this…

 

If I send you a dev. Board for this chip, would you test it for compatibility?

 

Thanks,

Rick

rbickle, 2020-02-21:
I have also found that if I erase the flash completely and flash only the bootloader code using the JTAG device, BOSSAC will program the application correctly at the offset. It will not verify or read the code back however.