shumatech/BOSSA #125: Flashing Arduino Due results in board not booting

sdroege, 2020-03-15:
Since ee5a651aacaf2c3fbcd539789574618a425d1a06. The previous commit 26154375 still works (after applying fbca22d36b351a6b771cd34a0b632d890c2db0ab, without this flashing fails directly).

Flashing works fine in either case, but the board is not booting up and does not run any code apparently (LEDs etc are in default state).

Let me know if you need any further information or if I can help debugging this somehow.

nzmichaelh, 2020-07-15:
Related: platformio/platform-atmelsam/issues/78 zephyrproject-rtos/zephyr/pull/26866

mike-kfed, 2021-02-01:
This issue seems to be fixed by either PR #111 or PR #151 for the 1.9 series of bossac

OptifySudarshanPatil, 2022-02-11:
Hi @sdroege ,
If you're using clone due have a look [here](https://copperhilltech.com/blog/arduino-due-design-flaw-due-wont-start-after-poweroffon-requires-reset/#:~:text=In%20detail%2C%20the%20Arduino%20Due,result%20in%20the%20described%20problem.)

iddq, 2023-01-11:
@sdroege any update?

janhohenheim, 2023-10-18:
@iddq I can confirm the issue still persists exactly as described on the newest version of BOSSA using an ATSAM3X8E on an Arduino due, just as @sdroege did.