shumatech/BOSSA #126: "Device unsupported" with Arduino Zero bootloader

lmagalhaes1337, 2020-03-25:
Hi there,

I'm trying to upload a code to an ATSAMD21G18 with the Arduino Zero bootloader using UART and BOSSA 1.9.1, however, after pressing reset twice, I always get the message "Device unsupported" in the GUI. Is this IC really unsupported or is there any other problem going on?

lmagalhaes1337, 2020-03-25:
Using the arduino IDE and choosing "Arduino Zero (Native USB)" I get this:

```
C:\Users\Luis\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3/bossac.exe -i -d --port=COM7 -U true -i -e -w -v C:\Users\Luis\AppData\Local\Temp\arduino_build_218381/Elite_1v2.ino.bin -R 
Set binary mode
No device found on COM7
readWord(addr=0)=0x30303077
readWord(addr=0xe000ed00)=0x30303030
readWord(addr=0x400e0740)=0x23342c30
version()=wE000ED00,4#w400E0740,4#V#
chipId=0x23342c30
Unsupported processor
Send auto-baud
Set binary mode
readWord(addr=0)=0x30303077
readWord(addr=0xe000ed00)=0x30303030
readWord(addr=0x400e0740)=0x23342c30
version()=wE000ED00,4#w400E0740,4#V#
chipId=0x23342c30
Unsupported processor
```

snehashakyas, 2022-07-21:
Did you happen to solve this error? I am getting the same error and I am very stuck. 

Your help would be much appreciated :)

Harsh-hit, 2022-08-22:
I am stuck with the same problem. Any help would be much appreciated :)

cozzyd, 2023-10-08:
This is very late, but I saw similar issues. Using BOSSA 1.7.0 worked in my case.