shumatech/BOSSA #136: Can only flash D21g18a once, then locked forever (nearly!)

soulsbysynths, 2020-07-01:
I have been following [this tutorial](https://microchipdeveloper.com/atstart:sam-d21-bootloader) for implimenting a bootloader for the D21g18a.  I've downloaded the relevant files, made my own device_config for it and successfully uploaded it.
I've then successfully uploaded the bootloader (via JTAG), built my binary (with flags to start at address 0x6000) and uploaded the binary using SAM-BA.  I've tested the upload via SAM-BA several times and seems to be fine.
I then did an upload via BOSSA using the windows GUI. It identified the D21x18 successfully and can show info. I set Flash Offset = 0x6000 and hit write.  All went well. 
I then tried again.  It said the flash was locked and asked to unlock first. I say yes, but it immediately brings up "Flash Command Failed".  I can still bring up info (it shows 0-15 locked regions) and read the flash.
I cannot write via SAM-BA either.  I get the error:
`send_file {Flash} "C:/Users/info/Documents/Atmel Studio/7.0/SsAtxController/SsAtxController/Debug/SsAtxController.bin" 0x06000 0
-I- Send File C:/Users/info/Documents/Atmel Studio/7.0/SsAtxController/SsAtxController/Debug/SsAtxController.bin at address 0x06000
File size : 142312
Flash size : 0x40000 Flash page size : 64 First page : 384
Dest : 24576
-I- 	Writing: 0x100 bytes at 0x6000 (buffer addr : 0x200028D8)
-E- Generic::Write returned error (0x00000002)`

I tried erasing the micro via JTAG and reuploading the bootloader and then the binary.  Still no joy (either SAM-BA or BOSSA). 

Weirdly, the only way to get the micro working is to set the upload project back to its default start address (is that 0x0000 ?), upload via JTAG.  Then start all over again (upload bootloader, change upload project start address etc etc).

I've been through this whole process twice, so definitely not a one-off. I'm a bit stuck and really want to use BOSSA in my project!  Any help would be much appreciated.

deladriere, 2020-07-01:
@soulsbysynths Hi Paul
I gave up on Bossa 
- For production, I use another SAMD21 that reads the .bin from an SD card and flash the target on a press of a button (see https://github.com/adafruit/Adafruit_DAP/tree/master/examples/flash_from_SD)
- Subsequent users updates are done via the SDU library (just by saving firmware.bin on an SD card and rebooting )
(see https://www.arduino.cc/en/Reference/SDU)
Hope this may help

soulsbysynths, 2020-07-01:
Ah that's a shame.  I feel that everything is so close to working with this system and with SAM-BA it's perfect, but of course I can't use that with my own uploader and I think it'll be too daunting for most users.  
I'd done a nice bootloader where you connect Clk Out to Clk In with a jack cable and it does a little handshake to put it into USB flash mode!
Will check out those links.  Flashing from SD is an option for me, so could go down that route.
I'm surprised that no-one else has run into this issue of BOSSA permanently locking the flash regions after writing.

thiagothimotti, 2021-05-20:
After this initial lock, you need to return SAMD21 fuse USER_WORD_1.NVMCTRL_REGION_LOCKS from 0x2000 to 0xFFFF.
Every new SAMD21 locks at the first time. You edit the Fuse and won't lock anymore.