shumatech/BOSSA #144: Steps to install BOSSA

rod6214, 2020-09-27:
Hi everyone, where do i find the instructions to build BOSSA repository?

beikeland, 2020-10-09:
there is also a make target for just the command line version that has less dependencies
`make bossac`

milesfrankland, 2021-01-04:
I couldnt get this setup to work.
This is the error:
APPLET WordCopyArm
make: ./appletgen: Command not found
make: *** [Makefile:252: src/WordCopyArm.cpp] Error 127

milesfrankland, 2021-01-05:
> I couldnt get this setup to work.
> This is the error:
> APPLET WordCopyArm
> make: ./appletgen: Command not found
> make: *** [Makefile:252: src/WordCopyArm.cpp] Error 127

I edited the files in windows and transferred to linux causing issues. When I git cloned from linux all worked well.

marts20, 2023-01-06:
I can't seem to compile for Windows. Can anyone compile with added support for D21G17D variant?

PeakeElectronicInnovation, 2023-03-09:
A couple of us managed to compile this in Windows and have documented the process on the Arduino forum in case it is helpful: https://forum.arduino.cc/t/creating-a-custom-samd-bootloader-and-arduino-board/1094588/10