shumatech/BOSSA #145: setting `Boot Flash` flag on ATSAM4SD32 does not always work

lowlander, 2020-10-02:
When I do a `bossac -e -w -v -b prog.bin`  the erasing and flashing work fine, but when I than do a `bossac -i` the `Boot Flash` is still `false`. Only if I do a second `bossas -b` the bit seems to be set (at that point bossac can't talk with the MCU anymore. A NRST of the CPU than boots my new `prog.bin` and everything works fine.

Am I doing something wrong or is there some timing/flash checking bug ?

bbeird1, 2021-11-15:
Hello, I am encountering the same situation with the ATSAM4S8B. Did you ever find a solution, or are you just setting boot from flash twice? Not a deal breaker at all, but I would like to know if there is operator error (wouldn't be the first time!) Thanks!

lowlander, 2021-11-15:
Hey Bob, no I never really found (nor did I actively look for) a solution. Also never got any reactions on this issue report, last code change also seems 2 years ago, it kind of looks like the bossa project is abandoned :-(

haata, 2021-11-15:
I think I've noticed this as well.
I suspect there's something going on with the chip.

See the datasheet errata: https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11100-32-bit%20Cortex-M4-Microcontroller-SAM4S_Datasheet.pdf 48.1.2 (Read Error after a GPNVM or Lock Bit Writing).