shumatech/BOSSA #148: Thanks!

bcr, 2020-12-28:
Not an issue, no action required. Nice clean repo, good work, builds great.

milesfrankland, 2021-01-04:
How on earth do you build?  I have tried to build in cygwin in windows - even created a new ubuntu instance to try and build in there.
Any pointers you can give me?

Thanks

milesfrankland, 2021-01-05:
> How on earth do you build? I have tried to build in cygwin in windows - even created a new ubuntu instance to try and build in there.
> Any pointers you can give me?
> 
> Thanks

I edited the files in windows and transferred to linux causing issues.  When I git cloned from linux all worked well.

bcr, 2021-01-06:
I'm on Linux, and I just did a `make bossac` and the magic happened.