shumatech/BOSSA #177: Can program SAMD5x on Windows, but not macOS

JetForMe, 2024-12-17:
I got a Teknic ClearCore motion controller that's based on an Atmel SAMD5x MCU, and they have Arduino support for programming it. I'm able to build the examples on macOS using Arduino 2.x, but programming the chip (using `bossac`) fails to verify. It does seem to write something, but it always fails to verify. Doing the same with Arduino on Windows works fine.

I emailed them over the weekend with a detailed message, but they just came back saying they don't support macOS.

Unfortunately, while Arduino on macOS shows the `bossac` invocation, it doesn't show the invocation or programming parameters on Windows.

I'm wondering if there’s something wrong with the way the macOS setup is invoking `bossac`:

```
"/Users/rmann/Library/Arduino15/packages/arduino/tools/bossac/1.9.1-arduino1/bossac" -i -d -a --port=cu.usbmodem3224101 -U -e -w -v --offset=0x4000 "/Users/rmann/Library/Caches/arduino/sketches/71F55F8C0907EEFE4D496BFF118C38F3/AsgWithMeasuredTorque.ino.bin" -R
```

The device seems to have a UF2 bootloader on it. When you connect it, a drive mounts with `CURRENT.UF2` and `INFO_UF2.TXT` which contains

```
UF2 Bootloader v2.3.1-Teknic.4 SLFHRO
Model: ClearCore
Board-ID: SAME53N19A-ClearCore-F1
```

The Windows invocation is identical, but it works. Any ideas why macOS might be failing?

RolfGgh, 2025-01-24:
@JetForMe 

I'm at the same situation using a Teknic controller.

The Arduino IDE install on macOS states that bossac version 1.9.1 is installed, but it's version 1.8

You can try this:
Go to
/Users/YOUR_USER_NAME/Library/Arduino15/packages/arduino/tools/bossac/1.9.1-arduino1
Exchange the bossac binary with the version 1.9.1 you can download here.

This works for me

JetForMe, 2025-01-25:
Yeah, that's what I did.

Deleavanlutheran, 2026-10-17:
"I"

Deleavanlutheran, 2026-10-17:
Ping

Deleavanlutheran, 2026-10-17:
172.56.7.222@://777

Deleavanlutheran, 2026-10-17:
10/4

Deleavanlutheran, 2026-10-17:
711

Deleavanlutheran, 2026-10-17:
10:40